EmotiBit/EmotiBit_Docs #126: Docs not giving correct info for MacOSX config file location

mikkokotila, 2024-08-10:
## Description

At least in MacOSX Sonoma, the current doc information fails to identify any means by which the config file can be found. The doc should make it very clear how and where to find it. 

## Type of correction
- [ ] Missing Content
- [ ] Broken Link
- [x] Inaccurate content
- [ ] Outdated/Deprecated content
- [ ] Inappropriate content
- [ ] Document structure needs updating

## Link
https://github.com/EmotiBit/EmotiBit_Docs/blob/master/Working_with_emotibit_data.md#emotibit-oscilloscope-settings

## Screenshots

nitin710, 2024-08-13:
@mikkokotila Can you highlight which config file you are talking about? The [link to config.txt](https://github.com/EmotiBit/EmotiBit_Docs/blob/master/Getting_Started.md#adding-wifi-credentials) is common for all operating systems.

And the location of the Oscilloscope Settings-files are [linked](https://github.com/EmotiBit/EmotiBit_Docs/blob/master/Working_with_emotibit_data.md#emotibit-oscilloscope-settings) in the location you have already highlighted. If you have specific comments on the problems you faced, please let us know and we can address those.

If you are talking about another configuration file, please let us know and we can update the documentation to increase the clarity.

nitin710, 2024-10-22:
@mikkokotila the application package is os agnostic.

For example, the docs say that the settings files can be found in `EmotiBitSoftware-macOS/EmotiBitOscilloscope.app/Contents/Resources/`
and that is where it is, even in the latest macos-Sequoia 15
[image]

Maybe the `.app` is causing the confusion? Macos by default hides the extension on application bundles, therefore `EmotiBitOscilloscope.app` shows up as `EmotiBitOscillosocpe`. If that is indeed the confusion you are highlighting I understand and we can make that patch to update to clarity.